ukmars/gemini #88: System level schematic

micromouseonline, 2025-09-06:
Is there a system level schematic or other diagram showing how the boards are interconnected and what the pins are?

Or did I just fail to find it?

If there is not, shall I do one?

davidhannaford, 2025-09-06:
Hi peter

Having another go at loading Quadratic

I swapped over the 2 jumpers on the discovery board and it programmed the chip on the discovery board with the quadratic code OK.

I didn’t have STM32Cube Programmer open when I did that

I removed the jumpers on the discovery board and tried to program quadratic again but I failed with :

Error: timed out while waiting for target halted

embedded:startup.tcl:1516: Error: ** Unable to reset target **

in procedure 'program'

in procedure 'program_error' called at file "embedded:startup.tcl", line 1553   

at file "embedded:startup.tcl", line 1516

 

I also tried the loadup with STM32Cube programmer open and connected and it failed with a different error:

Error: libusb_open() failed with LIBUSB_ERROR_ACCESS

Error: open failed

in procedure 'program'

** OpenOCD init failed ** 

 

Whatever has happened, now when I read the chip with STM32Cube programmer it is now showing all Hex FFs so the program that was in the chip has now gone. 

 

Should I need to have STM32Cube programmer open and connected when I do the programming ?

 

Any thoughts or ideas welcomed

Cheers

David

 

This email has been checked for viruses by AVG antivirus software.
www.avg.com

IanSuper7, 2025-09-07:
No diagram but a cross reference list in the ecad [folder](https://github.com/ukmars/gemini/blob/main/ecad/GeminiV2PicoPins.pdf)

micromouseonline, 2025-09-07:
I will make a picture.